Monash-Connected-Autonomous-Vehicle/ESDA #10: Implement safety light

AbBaSaMo, 2024-01-08:
- [ ] safety light must be on when the vehicle is powered on
- [ ] safety light must flash when the vehicle is in autonomous mode

AbBaSaMo, 2024-03-17:
Speak with Jai/mechanical on considerations or ideas they have with regards to the stack communicating with the light etc

Jiawei-Liao, 2024-03-21:
Implemented a timer to send messages to can at set intervals as long as a topic message is received.
For future topics, add "self.safety_light_timer_counter = 0" to the subscription callback.

Features awaiting electrical or some other team:
- Redundancy cycles (if any) so the signal continues to be sent even if no topic messages are received.
- Frequency of sending message.
- ID and value of message to be sent.

AbBaSaMo, 2024-03-25:
@shrayb am going to assign you to this as I believe you'll be able to help Jiawei speed up the process of determining these things from electrical and in general. Can the both of you try get this fully done by the end of this week and push elec to get their part done/questions answered