Monash-Connected-Autonomous-Vehicle/ESDA #12: Implement navigation2 costmap pothole layer

AbBaSaMo, 2024-01-08:


AnthonyZhOon, 2024-02-15:
I think this is not directly writing a costmap 2d layer, it seems like we can make another occupancy_grid_camera topic and add it with a occupancy grid generated by lidar.  This might require writing an occupany grid publisher for both lidar and camera sensors. Then the costmap is generated the same for both lidar and camera obstacles.

AbBaSaMo, 2024-02-15:
@AnthonyZhOon hmm, so do you mean rather than create costmap layers, we fuse our multiple occupancy grids and publish the result to the /map topic which nav2 uses?

So fuse:
- slam output
- lane detection
- pothole detection

AnthonyZhOon, 2024-02-15:
I think it makes more sense to identify all obstacles there as the costmap is a postprocessing on these obstacles to create a heatmap. though there is an argument that only hard walls are lethal obstacles and lanes can be weaker obstacles

AbBaSaMo, 2024-02-17:
@AnthonyZhOon makes sense. In that case I'll update the issues to reflect this as generating occ grids to be fused.

AbBaSaMo, 2024-02-17:
Actually gonna close the issue rather than delete as this can all be reflected and discussed in this one https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/issues/25 and 2 is now redundant rather than 1

AbBaSaMo, 2024-02-17:
Re-opening issue and re-purposing it to fusion of the 3 occupancy grids:

- [ ] have a topic potholes/map
- [ ] have a topic lanes/map
- [ ] have a topic remapped from slam_tb slam/map
- [ ] fuse the 3 into topic /map

AnthonyZhOon, 2024-03-24:
Written a node for fusing 2 occupancy grids in the occupancy-fuser branch. Need to test it still but had trouble showing grids in rviz2.
It creates a new grid based on the largest width and height of collected grids, and uses a 2d convolve to add each subscribed grid to the centre of the new grid.

AbBaSaMo, 2024-03-24:
@AnthonyZhOon any errors being output by rviz or nodes or screenshots? could it be a missing transform issue?

I can take a look monday after 4. Either way moving issue from in progress to review.

AbBaSaMo, 2024-03-25:
@AnthonyZhOon are there unpushed changes you've made? colcon build outputs the following error when attempting to buils occupnacy-fuser

[image]

Use of hyphen is said to be the issue as per https://answers.ros.org/question/368144/colcon-build-fails-on-newly-created-package/ but I don't want to trial it given you seem to have gotten it built and I might just be missing something.

Rosdeps have al been installed

AbBaSaMo, 2024-03-25:
@AnthonyZhOon changed package name to using '_' insread of '-' which fixed the issue. If I can confirm you have no unpushed changed, am gonna push.

[image]

AbBaSaMo, 2024-03-25:
It seems the issue is that the occupancy grids have not been initialized with correct parameters, so based on the following grid1 message, you have a 51x51 grid with only cell 50 occupied, and a resolution of 0 so each cell is 0x0m iirc

[image]

AbBaSaMo, 2024-03-25:
@AnthonyZhOon can confirm that this outputs *a* grid. will push after class plus a meeting, and possible add a random generator for cells

[image]

AbBaSaMo, 2024-03-25:
Another thing to note is that you'll probs need a static tf publisher for base_link. Publishing an arbitrary one, but I'll have to confirm if it doesnt work without it

[image]

AbBaSaMo, 2024-03-26:
@AnthonyZhOon pushed fixes and changes, so the remaining tasks for testing are:
- [ ] fix grid publishers so they aren't occupied at just cell 50, make it random or smth
- [ ] test and verify accuracy of fusion
- [ ] consider that the resolution of the grids might differ