Monash-Connected-Autonomous-Vehicle/ESDA #18: navigation2 configuration

AbBaSaMo, 2024-01-08:


AnthonyZhOon, 2024-02-15:
Using the `simulation_launch.bash` in ~/esda_igvc on hive 5 will now launch the gazebo sim + nav2 stack. Still need to figure out configuration for ESDA, stripping unnecessary parts

AnthonyZhOon, 2024-02-15:
@AbBaSaMo What parts do you think are replaced by `robot_localisation`, I know the amcl_server node in nav2 manages sensor fusion with a particle filter and is probably replaced

AbBaSaMo, 2024-02-15:
Hmm, not sure what is replaces per say but what it provides is:

- sensor fusion as you had already mentioned
- map -> odom -> base_link transform which I'm note sure if any other component of nav2 provides
- is implicitly suggested to be used by the nav2 wiki if we want to use GPS based localization/follow GPS waypoints https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html

closed 2024-07-20 by AbBaSaMo