Monash-Connected-Autonomous-Vehicle/ESDA #2: Create ESDA xacro/urdf

AbBaSaMo, 2024-01-08:
Create a robot model for the ESDA with sensors and steering geometry accurately modelled.

AbBaSaMo, 2024-02-01:
Found this tool via a youtube vid I can use https://onshape-to-robot.readthedocs.io/en/latest/index.html

AbBaSaMo, 2024-02-13:
This link provides some instructions on including an IMU for a gazebo sim https://navigation.ros.org/setup_guides/odom/setup_odom.html
This can be used for gps http://wiki.ros.org/hector_gazebo_plugins

AbBaSaMo, 2024-02-13:
will try to use this tool on the ubuntu laptop https://github.com/Rhoban/onshape-to-robot
https://www.youtube.com/watch?v=C8oK4uUmbRw
https://www.youtube.com/watch?v=TJeCpGnX508

AbBaSaMo, 2024-02-24:
### onshape-to-robot (o2r) usage
Install o2r using `pip install onshape-to-robot`

Go to https://dev-portal.onshape.com/keys to create an API key
- obtain access key 
- obtain secret key

Document ID is the string in the URL after .com/documents/ so `42045ec8cb8376f89673d4c1`

Create a new directory and then create config.json inside it 
```
mkdir my_robot_dir
cd my_robot_dir
touch config.json
```

Write up the config.json file with the following contents
```
{
    // ideally store these 3 as ENV vars else include in config.json as shown
    "onshape_api": "https://cad.onshape.com",
    "onshape_access_key": "[THE ACCESS KEY GENERATED]",
    "onshape_secret_key": "[THE SECRET KEY GENERATED]",

    "documentId": "42045ec8cb8376f89673d4c1", // obtain as explained above
    "assemblyName": "Asterius", // must specify else the first assembly found will be used
    "outputFormat": "urdf" // can also be sdf if needed
}
```

Refer to https://onshape-to-robot.readthedocs.io/en/latest/config.html for more configurations that can be set

Run the following command to import the robot as urdf `onshape-to-robot path_to_config_dir`

AbBaSaMo, 2024-02-24:
@AnthonyZhOon the result is a URDF with no indentations, 17 000 lines long (yes 17k) that will be too complex to modify, add sensors to etc. I don't think using cad to urdf tools is feasible given the complexity of the model. I'll have to either ask mechanical to create a really simplified cad else work with them during a working session to make a urdf from scratch.

AbBaSaMo, 2024-02-24:
According to ROS standards, root link in urdf must be base_link with orientation:
- x-forwards
- y-left
- z-up

Pre-requisites for this build 
- xacro
- joint state publisher + gui

xacro processes urdf/xacros into a single xacro that robot_state_publisher processes

include another xacro file usinf
`xacro:include filename="other file's name" />`

the general way to create links and joints is

```
<link name="parent_name">
    <!-- defines what we see -->
    <visual>
        <origin xyz="x y z"/>
        <geometry><!-- shapes go here --></geometry>
        <material name="name if predefined here" />
    </visual>

    <!-- defines the physics/collision box -->
    <collision>
        <origin xyz="x y z"/>
        <geometry><!-- shapes go here --></geometry>
        <material name="name if predefined here" />
    </collision>

    <!-- also need to define physics in terms of inertia and stuff -->
    <intertia></inertia>
</link>

<joint name="parent_child_joint">
</joint>

<link name="child_name">
    <!-- child link elements here relative to parent -->
</link>
```

AbBaSaMo, 2024-03-05:
Reminder to self:
- non fixed joints need robot state publisher gui running when testing to display

AbBaSaMo, 2024-03-05:
run 

```
colcon build --packages-select esda_sim; source install/setup.bash; ros2 launch esda_sim rsp.launch.py use_sim_time:=true;

ros2 run joint_state_publisher_gui joint_state_publisher_gui

ros2 launch gazebo_ros gazebo.launch.py
```

to spawn robot in gazebo
`ros2 run gazebo spawn_entity.py -topic robot_description -entity my_robot_name`

AbBaSaMo, 2024-03-05:
refer to https://youtu.be/IjFcr5r0nMs?list=PLunhqkrRNRhYAffV8JDiFOatQXuU-NnxT&t=1210 on spawnign in a cusotm built world

AbBaSaMo, 2024-03-05:
when launching sim launch, add the path to the world as the value to param world:=

AbBaSaMo, 2024-03-05:
@AnthonyZhOon simulated lidar has been added, just need to add camera now and then imu/gnss

good new is the zed wrapper repo we use provides a urdf we can import to our own https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed_descr.urdf.xacro

[image]

AbBaSaMo, 2024-03-05:
Have got the zed-camera urdf mounted in sim 

[image]

ONly issue is that it doesn't load in gazebo --> it waits for the /spawn_entity service which takes it's time and stops and also in rviz2 it has some error Could not load mesh resource 'package://zed_interfaces/meshes/zed2.stl'

@Monash-Connected-Autonomous-Vehicle/avp-team anyone able to help?

dylan-gonzalez, 2024-03-06:
If you want some nodes to only launch when another node has launched, you could do something like [this](https://github.com/ros-controls/ros2_control_demos/blob/d472f5ea4f67f8c81051ed6297528872b59e85e2/example_10/bringup/launch/rrbot.launch.py#L82-L87)

For your `package://zed_interfaces/meshes/zed2.stl` error, maybe try `file:///$(find zed_interfaces)/meshes/zed2.stl` ?
Also for future reference:

> When you're looking for a file using file:// like http:// usually you need 3 slashes to tell it that it's a absolute url. try "file:///home/username/path/path/filename.stl" if you don't the first argument after the slashes is considered the hostname.
https://answers.ros.org/question/39116/bug-mesh-file-path-in-urdf-cannot-locate-resource-file/

AbBaSaMo, 2024-03-06:
@dylan-gonzalez pretty sure i tried the find syntax but that didn't work, this isn't my code but stereolabs own code, i'll try the 3 slashes and see if it works tho

dylan-gonzalez, 2024-03-06:
If you're using your local copy of zed_interfaces make sure it's built

I'm also not too sure what you mean about the URDF loading issue

AbBaSaMo, 2024-03-07:
@dylan-gonzalez so everything is built, it's in the right location in install, but when the XACRO tries to load the mesh for the camera, it seems like it's able to find it but it just can't load and render it. This is in RVIZ2

When I run my launch file that runs gazebo, robot_description and the spawn_entity service [which spawns the robot_description into gazebo], the service runs for a bit but cannot sapwn the entity so the model does not render/spawn in gazebo. I think this is related to the mesh issue.

AbBaSaMo, 2024-03-07:
@Monash-Connected-Autonomous-Vehicle/avp-team 

https://github.com/orgs/Monash-Connected-Autonomous-Vehicle/projects/11/views/1?pane=issue&itemId=49985426 all messages from the swiftnav driver are under the same frame id

im not sure if i need to model the piksi multi in the urdf to have the transform between it and baselink. and if I do, idk if I need to model the GPS antenna as a separate link to the piksi itself.

robot localization provides the map > odom > baselink transform using twist, gnss and imu in case that would help make the decision

AbBaSaMo, 2024-03-07:
putting piksi dimensions as 12 x 14 x 4 which is what is used in the cad

AbBaSaMo, 2024-03-07:
I need to refer to this for orientation
https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/issues/19#issuecomment-1940490283

AbBaSaMo, 2024-03-07:
https://github.com/fsuarez6/labrob/blob/master/labrob_description/urdf/labrob.urdf.xacro#L202
https://github.com/qboticslabs/ros_robotics_projects/blob/master/chapter_10_codes/sensor_sim_gazebo/urdf/gps.xacro

AbBaSaMo, 2024-03-08:
re did the wheels and body with jason, we're gonna see if we can redo the wheels as ackerman but have 1 middle turning wheel and 2 static ones that are frictionless [casters]

[image]

AbBaSaMo, 2024-03-12:
https://stackoverflow.com/questions/77634282/how-to-use-ackermann-plugin-in-ros2-humble
https://robotics.stackexchange.com/questions/105969/how-to-use-gazebo-ackermann-plugin-ros2-humble
https://answers.ros.org/question/323637/is-there-a-ros-example-for-an-ackermann-vehicle-sim-that-uses-the-ackermann_steering_controller-and-the-corresponding-gazebo-plugin/

AbBaSaMo, 2024-03-15:
I've managed to get the ESDA launching while using the ackermann steering plugin, but I can only get the front wheels to rotate around one axis and not both axis they need to rotate about. Teleop twist keyboard also no longer causes the robot to move. Have pushed to git, link to commit is here https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/tree/b2a3203d91382fb9e723d037ca9b2643347a1aef in case I make changes or smth, if anyone can look into ESDA/esda_sim/description to help debug, would be much appreciated.

[image]

It seems, that the issue is that there are 2 joints in steering_wheels.xacro defined, 1 to enable them to rotate around z axis, another for rotation around x so both steerign and rolling motion. The issue is idk how to properly implement this and there's not much info online.

AbBaSaMo, 2024-03-15:
I think this might be a solution to my issue, https://answers.ros.org/question/341365/urdf-attaching-two-joints-to-one-link-for-a-gripper/ yet to implement it

AbBaSaMo, 2024-03-15:
Mimic joints was not the issue. The problem is that I need 1 link (the wheel) to spin on 2 axis, X and Z to model the front wheels but there does not appear to be a way to do this.

Creating 2 joints with the same parent and child causes the issue. Having the joint with different parent causes a cyclic link tree which is not allowed. I made 2 front left and 2 front right wheels but this will likely have physics and collision box issues. 

These look promising though
https://robotics.stackexchange.com/questions/104023/ackermann-plugin-ros2-foxy-how-to-use-it

This too
https://robotics.stackexchange.com/questions/105969/how-to-use-gazebo-ackermann-plugin-ros2-humble

AbBaSaMo, 2024-03-15:
Looking at gazebo example world, they use a single joint for front wheels that is a sphere and supply that as both steer and front wheel joints

[image]

This is the left wheel for example

```
      <link name="front_left_wheel">
        <pose>1.41 0.76 0.3 0 0 1.57</pose>
        <inertial>
          <mass>11</mass>
          <inertia>
            <ixx>0.58631238</ixx>
            <ixy>0.0</ixy>
            <iyy>0.33552910</iyy>
            <ixz>0.0</ixz>
            <iyz>0.0</iyz>
            <izz>0.33552910</izz>
          </inertia>
        </inertial>

        <visual name="visual">
          <pose>0.04 0.0 0.0 0 0 0</pose>
          <geometry>
            <mesh>
              <uri>model://prius_hybrid/meshes/Hybrid.obj</uri>
              <scale>0.01 0.01 0.01</scale>
              <submesh>
                <name>Wheel_Front_Left_</name>
                <center>true</center>
              </submesh>
            </mesh>
          </geometry>
        </visual>

        <collision name="collision">
          <pose>0.0 0.0 0.0 0 1.5707963267948966 0</pose>
          <geometry>
            <sphere>
              <radius>0.31265</radius>
            </sphere>
          </geometry>
          <surface>
            <friction>
              <ode>
                <mu>0.9</mu>
                <mu2>0.9</mu2>
                <slip1>0.0</slip1>
                <slip2>0.0</slip2>
              </ode>
            </friction>
            <contact>
              <ode>
                <min_depth>0.001</min_depth>
                <kp>1e9</kp>
              </ode>
            </contact>
          </surface>

        </collision>
      </link>
```
and this is the joint

```
      <joint type="universal" name="front_left_combined_joint">
        <pose>0 0 0 -0.08726646259971647 0 0</pose>
        <child>front_left_wheel</child>
        <parent>chassis</parent>
        <axis>
          <xyz>0 0 1</xyz>
          <limit>
            <!-- +- 50 degrees -->
            <lower>-0.8727</lower>
            <upper>0.8727</upper>
          </limit>
        </axis>
        <axis2>
          <xyz>1 0 0</xyz>
          <dynamics>
            <friction>18.0474092253</friction>
          </dynamics>
        </axis2>
      </joint>
```

Though sdf allows 2 axis of rotation

AbBaSaMo, 2024-03-15:
Managed to implement wheels turning on 2 axis. The trick is adding a link that turns on one axis and then attaching a wheel to it but the wheel turns on another axis.

[image]

Now the issue that remains is why the ESDA won't render on Gazebo and why the Ackermann Plugin cannot find the joints I provide it despite them being shown in RVIZ2

[image]

AbBaSaMo, 2024-03-15:
Not it loads but the wheels are oriented on base_link. No error outputs in terminal

[image]

AbBaSaMo, 2024-03-15:
And here's the tf tree

[image]

SO to summarise
- esda is not rendering in Gazebo 
- esda is not moving in Gazebo
- all wheels are squashed on baselink

BUT
- transforms are fine
- logs indicate the plugin is working fine as well as the urdf is fine
- without the plugin i.e. commenting it out, everything works find though the esda does very slowly roll backwards in sim
- under either scenarios it works in rviz

The plugin in question is https://docs.ros.org/en/ros2_packages/rolling/api/gazebo_plugins/generated/classgazebo__plugins_1_1GazeboRosAckermannDrive.html#exhale-class-classgazebo-plugins-1-1gazeborosackermanndrive

@AnthonyZhOon @dylan-gonzalez any ideas?

dylan-gonzalez, 2024-03-15:
I'd have to take a look at it myself

I might be able to see on Monday

AbBaSaMo, 2024-03-16:
@dylan-gonzalez yep, but available after 4pm

AbBaSaMo, 2024-04-18:
As of now the URDF looks as such 

[image]

And the following remain to be completed
- [ ] add the camera
- [ ] add a plugin for IMU and GPS in gazebo

AbBaSaMo, 2024-04-23:
Camera position has been added. IMU/GPS plugins etc are going to be moved to a separate sim task and this will only revolve around the tf tree provided by the urdf

closed 2024-04-23 by AbBaSaMo