Monash-Connected-Autonomous-Vehicle/ESDA #3: Create IGVC course xacro/urdf

AbBaSaMo, 2024-01-08:
Create a model of the IGVC course including mock obstacles and ramp.

AbBaSaMo, 2024-02-17:
@AnthonyZhOon just confirming that you decided to take this one?

AbBaSaMo, 2024-02-20:
@AnthonyZhOon we can reuse this fork of ours and the racecourse in it https://github.com/Monash-Connected-Autonomous-Vehicle/minidrone_linorobot/tree/foxy

launch https://github.com/Monash-Connected-Autonomous-Vehicle/minidrone_linorobot/blob/foxy/linorobot2_gazebo/launch/minidrone_sim.launch.py

the map it uses https://github.com/Monash-Connected-Autonomous-Vehicle/minidrone_linorobot/blob/foxy/linorobot2_gazebo/worlds/map1.txt

would just need to add potholes and the ramp iirc and to use humble instead of the foxy branch from the source repo if it has one

AnthonyZhOon, 2024-02-22:
The linorobot map is usable, we can try disabling physics of the ground plane to see if it works better.

closed 2024-07-20 by AbBaSaMo