Monash-Connected-Autonomous-Vehicle/ESDA #5: Implement motor interface for ros2_control

AbBaSaMo, 2024-01-08:
Refer to the following docs https://control.ros.org/humble/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html

AbBaSaMo, 2024-01-27:
Refer to the following diagram on how things work 

[image]

So you guys will now write 1 interface that communicates to the MCU chip via can and reads velocity/wheel position data and writes motor commands. The MCU computes velocity and wheel position using the encoder data and electrical will set that up.

Further details will be discussed this upcoming Wednesday.

AbBaSaMo, 2024-02-17:
@AnthonyZhOon this is a milestone 2 task but moving it to 'in progress' as https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/issues/7 is in testing/review phase and other tasks cannot be worked on by will/jiawei due to all ready being assigned or blocked by something

elec lead said this usb to can device is good to go with but we need to purchase it before the latter tasks of this issue can be worked on https://core-electronics.com.au/usb-to-can-adapter-model-a-stm32-chip-solution-multiple-working-modes-multi-system-compatible.html 

I might do so next week but will update if otherwise

Jokua, 2024-02-19:
I've managed to get the node up and running that subscribes to the topic cmd_vel and display only the linear.x msg of the cmd_vel twist message. 

[image]

As a note for myself, when adding a new node to the esda_control package, you have to make a new python executable in the package folder, and once that is complete, go into the setup.py and add a line to the "entry_points" console scripts and update it there to work with ```ros2 run```. It dosen't stop there as we have to update the launch file as well to start up the new node.

Not entirely sure how to convert it to the CAN message, apart from the [link](https://github.com/GOFIRST-Robotics/ros2socketcan_bridge/tree/master) Jiawei sent me. Does anyone from @Monash-Connected-Autonomous-Vehicle/avp-team know what I can use for the CAN messages bit? Also either walk me through or suggest some reading material to look through. Thanks

AbBaSaMo, 2024-02-19:
@Jokua nice, push to the repo on the same (esda-control) branch and make sure you pull before each push cause I'm making some changes to the launch file.

Dw about the CAN stuff atm cause we're waiting for our usb to can device to come before we look into requirements for using it such as suitable packages etc.

AbBaSaMo, 2024-02-19:
@Jokua nm actually, if you want, you can look into how to send messages to this device https://core-electronics.com.au/usb-to-can-adapter-model-a-stm32-chip-solution-multiple-working-modes-multi-system-compatible.html?gad_source=1&gclid=EAIaIQobChMIpfiu-bC2hAMVH6lmAh2XDglYEAQYASABEgK0Q_D_BwE in the appropriate can format we discussed with Jai and Callum, since this is the device we'll be using. We just can't test things until it gets here.

Jokua, 2024-02-19:
ok, i've pulled first to make sure there's nothing updated yet, and i've just pushed my changes @AbBaSaMo

Jokua, 2024-02-19:
ok, i'll take a look

AbBaSaMo, 2024-02-19:
Small update to the node and clarifying instructions added

[image]

So we aren't making a publisher, but we're connecting to a serial device, then whenever we get a linear vel, we convert it to a can msg and send it via that serial device i.e. the one I linked above.

Jiawei-Liao, 2024-02-21:
I've added code for connecting to serial (although not sure on what serial port and baud rate to use) and creating and writing CAN message. Not sure what is meant by the note on line 24.

I think default baud is 2000000 as stated by the website linked above?

Also please have a look at it because I'm not too sure if its correct.

AbBaSaMo, 2024-02-21:
What's meant is that we send 2 separate commands for the 2 wheels: 1 for left wheel, 1 for right wheel e.g.

cmd_vel says go at 9m/s, we tell left wheel go 9m/s and then send a 2nd command at the same time to right wheel telling it to go 9m/s

AbBaSaMo, 2024-02-24:
@Jiawei-Liao am getting this error launching the control module 

[image]

had you encountered this? also is it appropriate to use given we're using a custom can msgs design? my impression was that we just create a binary var and set the different bytes in accordance with the design callum and jai went over

AbBaSaMo, 2024-02-24:
Tried adding ros_canopen as well as ros2_canopen to package.xml but rosdep install cannot find the packages even after apt-get update
@Jiawei-Liao if it's needed to use this package we'll have to add the repo as a submodule else use something else.

Jiawei-Liao, 2024-02-24:
Totally forgot we were using custom messages so maybe it’s not needed. If it’s not working, would it be possible to just copy and paste the frame message?

AbBaSaMo, 2024-02-24:
Wdym by copy paste the frame message?

Jiawei-Liao, 2024-02-24:
Like just copying the message into the folder

Jiawei-Liao, 2024-02-24:
For the custom can message, where was that image for its layout? And did you want it to be implemented here or some other folder?

AbBaSaMo, 2024-02-24:
@dylan-gonzalez just confirming as long as we have a binary var with the following structure, we're good to go and just need to send it via serial. So e.g. we send the bits 01 000001 0000110 representing a command over serial to the usb-to-can converter and the converter will do the rest without needing to use any can libraries?

[image]

AbBaSaMo, 2024-02-24:
@Jiawei-Liao adding link to the notion with the layout to the issue description

dylan-gonzalez, 2024-02-24:
im not sure what converter you're using... probably a question for the elec team

AbBaSaMo, 2024-02-27:
@Jiawei-Liao @Jokua keep in mind the servo is using USB0 so when we connect to serial for this device, we can't use the same thing

Jiawei-Liao, 2024-02-27:
Do we know if it'll be on another one like USB1 or something? But I'll keep that in mind to change it after tomorrow's meeting

Jiawei-Liao, 2024-03-06:
I think sending floats mostly works. Although not sure why there is a small difference in the received data and transmitted data. Maybe there is too much precision in the send data?

[image]

[image]

This is code on the CAN bus/Arduino side. I have given it 8 bytes of buffer which should be able to store a float64? So I'm not sure how to fix this rounding error.

[image]

AnthonyZhOon, 2024-03-06:
looks like it is missing bits near the last place, sign, exponent and
leading mantissa are the same due to cutoff. You might need more than
exactly 8 bytes of buffer due to length encoding, try printing the raw
bytes on both sides to see?

Jiawei-Liao, 2024-03-06:
I changed it to double from single so that it can take a larger precision now. Printing out the bytes and value to 12 dp on both sides shows the same values. I think the extra decimal places is just a limitation of floats and doubles. 

[image]

[image]

AbBaSaMo, 2024-04-24:
@Jokua @Jiawei-Liao 

What is the state of this code and its branch? Is it fully complete? Last I remember we wanted to test it. I want to know if it can be merged or not.

Jiawei-Liao, 2024-04-24:
I made a final commit to allow sending messages using the correct IDs for left and right wheels. I accidentally added some zed camera commits to this branch as well. But I think they'll be overwritten once the zed camera branch is merged. I'm not sure if it has been added to control launch py. Not sure where that is located

Other than that, it can be merged now.

Jiawei-Liao, 2024-04-25:
How do I merge this branch?

AbBaSaMo, 2024-04-25:
@Jiawei-Liao what time are you in tomorrow? we'll do it then and have a quick code review

Jiawei-Liao, 2024-04-26:
yep I'm here now

AbBaSaMo, 2024-05-01:
marking as closed, testing will ne new issue

closed 2024-05-01 by AbBaSaMo