Monash-Connected-Autonomous-Vehicle/ESDA #6: Implement servo interface for ros2_control

AbBaSaMo, 2024-01-08:
Refer to the following docs https://control.ros.org/humble/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html

Jiawei-Liao, 2024-01-26:
Following instructions from here:
https://github.com/dynamixel-community/dynamixel_hardware/blob/humble/README.md

The git clone git command was changed from using this:
https://github.com/youtalk/dynamixel_control.git
To this:
https://github.com/dynamixel-community/dynamixel_hardware.git
Because it did not contain the .repos file used next.

This line was changed from this:
vcs import src < src/dynamixel_control.repos
To this:
vcs import src < src/dynamixel_hardware/dynamixel_control.repos

This built successfully except with some warnings.
[image]

Using this command in one terminal seems somewhat successful. A lot of text was outputted. There was an error for failing to open port since the servo usb did not connect. (Probably because the charger used was rated 12V 3A while it requires 12V 7A according to notion)

It opened up rviz.
[image]

Continuing with the rest of the command listed on another terminal: 
The command:
ros2 control switch_controllers --activate joint_state_broadcaster --activate joint_trajectory_controller --deactivate velocity_controller
Gave this error. Most likely due to the previously mentioned usb error
[image]

The command:
ros2 action send_goal /joint_trajectory_controller/follow_joint_trajectory control_msgs/action/FollowJointTrajectory -f "{
  trajectory: {
    joint_names: [joint1, joint2, joint3, joint4, gripper],
    points: [
      { positions: [0.1, 0.1, 0.1, 0.1, 0], time_from_start: { sec: 2 } },
      { positions: [-0.1, -0.1, -0.1, -0.1, 0], time_from_start: { sec: 4 } },
      { positions: [0, 0, 0, 0, 0], time_from_start: { sec: 6 } }
    ]
  }
}"
Gave this error, waiting forever.
[image]

This interface is probably hopefully done? Just need to test with working servo.

Jiawei-Liao, 2024-02-01:
Apparently usb connection does not work for wsl2. So we re-installed it onto the "car's" computer - hive-5.

The community version built but the run commands did not work. 

Instead, we followed the instructions from here:
https://www.youtube.com/watch?v=E8XPqDjof4U

Built successfully and ran, allowing us to read and write to the servo using these commands.
Write:
ros2 topic pub -1 /set_position dynamixel_sdk_custom_interfaces/msg/SetPosition "{id: 1, position: 4000}"
The position can be set to anything between 0 and 4000.
Read:
ros2 service call /get_position dynamixel_sdk_custom_interfaces/srv/GetPosition "id: 1"
For some reason, the read position is kind of incoonsistent. When testing with position 0, it initially gave 37. Later after more testing, it gave 9. When testing with position 4000, it initially gave 3997. Later, after more testing, it gave 3999.
Not sure why there are these inconsistencies and how to fix them.