Monash-Connected-Autonomous-Vehicle/ESDA #22: Z Camera driver and launch

AbBaSaMo, 2024-01-15:
- [ ] Get the Z Camera working and outputting an image via rviz
- [ ] Where applicable, get a launch file with any required parameters/configurations implemented so we can launch this using our [overall] sensor kit launch file

Refer to the following repository https://github.com/stereolabs/zed-ros2-wrapper
Refer to the following issue in the AVP project which also deals with setting up the Z Cam https://github.com/Monash-Connected-Autonomous-Vehicle/ENV200-Shadow-Repo/issues/4

m076em, 2024-01-18:
Z cam setup:
1-Install [CUDA](https://developer.nvidia.com/cuda-downloads)
2-Download corresponding version of Zed [SDK](https://www.stereolabs.com/developers/release#82af3640d775)
3-[Install SDK](https://www.stereolabs.com/docs/installation/linux)

AbBaSaMo, 2024-01-25:
@m076em since you mentioned the drivers work with no issues, and based on the driver README there doesn't seem to be any modifications to any launch/config files, I'll close this issue.

AbBaSaMo, 2024-01-25:
@m076em would you happen to know the model of the zcam?

[image]

AbBaSaMo, 2024-01-28:
Experienced the following error from zed pkgs while trying to build:

[image]

Tried installing `sudo apt-get install libjpeg-turbo8-dev libturbojpeg` which solved the issue though it's odd that I did not experience it before

[image]

m076em, 2024-01-28:
I haven't seen this error either, it might do with a incompatible version of the sdk driver, but I am not sure

AbBaSaMo, 2024-03-17:
@Jokua @AbBaSaMo update stack so it uses the zed2i

AbBaSaMo, 2024-03-19:
updated launch arg to zed2i

closed 2024-03-19 by AbBaSaMo