Monash-Connected-Autonomous-Vehicle/ESDA #23: Swift Navigation kit driver and launch

AbBaSaMo, 2024-01-15:
- [ ] Get the nav unit working and outputting odomery, imu and gps
- [ ] Where applicable, get a launch file with any required parameters/configurations implemented so we can launch this using our [overall] sensor kit launch file

Refer to the following repository https://github.com/swift-nav/swiftnav-ros2

AbBaSaMo, 2024-01-23:
Following resources are for the swift piksi multi [i.e. what I've been referring to as the swift nav kit]

[Overview of the piksi](https://www.swiftnav.com/sites/default/files/piksi_multi_product_summary.pdf)
[Product specs](https://www.swiftnav.com/resource-files/Piksi%20Multi/v2.2/Specification/Piksi%20Multi%20HW%20Specification%20v2.2%20[000-534-02-02].pdf)

[The evaluation board [the big blue one on the bottom](https://support.swiftnav.com/support/solutions/articles/44001850756-piksi-multi-evaluation-board)

[Using the piksi](https://support.swiftnav.com/support/solutions/articles/44001850752-piksi-multi-getting-started-guide)

[Using ROS with the Piksi](https://support.swiftnav.com/support/solutions/articles/44001907919-using-ros-with-swift-navigation-gnss-devices)

AbBaSaMo, 2024-01-23:
- [x] installing swift console from https://www.swiftnav.com/resource-library?filters=no&title=Swift+Console&search=Swift+Console&product=Swift+Console&category=Installer&release=Latest and then running dpkg -i /path_to_.deb
- [x] linux already has the specific USB to serial drivers https://support.swiftnav.com/support/solutions/articles/44001903704-installing-usb-to-serial-adapter-drivers
- [x] connect power, usb to serial [note it HAS to be RS2321] and antenna

Jiawei-Liao, 2024-01-23:
@dylan-gonzalez do you know how to get the lidar ethernet connection to work? I've tried plugging it into my own laptop but it just says ethernet unplugged and not connected. I've followed the instructions on notion and the manual, changing the ip address and subnet mask but it doesn't help.

AbBaSaMo, 2024-01-23:
No issues running swifts 1st party driver. The only change in config file is that we specify the device communication type as USB/serial which is a value of 2

[image]

Need to test outside where the antenna can receive GPS signals though as only IMU is priovided and the other message types are not outputting.

AbBaSaMo, 2024-01-27:
Navsatfix is now publishing as well as twistwithcovariancestamped. 

[image]

[image]

The issue was the set of sbp messages the piksi multi was configured to send. We need imu to be set as true and the sbp messages as outlined in the driver repo readme to be specified in the UART1 settings.

Having issues setting this to specific sbp messages though. Only default and no entry [which means all types] can be saved. Maybe I'm not formatting it correctly. Eitherway, blank works.

closed 2024-01-27 by AbBaSaMo