Monash-Connected-Autonomous-Vehicle/ESDA #39: Get support from servo supplier regarding packet handler issue

AbBaSaMo, 2024-03-26:
When we hook up the servo and run the read_write_node that allows us to control it, the node itself works, but the packets being sent seem to have some issue. Why? Because:
- the node is the original source code provided by dynamixel that we have always been using [and is unchanged]
- the node previously worked
- the node does not output any node related errors i.e. the ros2 service to set the position returns correctly
- now it only doesn't move the servo and logs the following error:

TODO
- [ ] email servo supplier for support
- [ ] handle suggestions from supplier

shrayb, 2024-04-11:
Tribotix guy asked us to run the wizard. From doing that we get: 

[image]

AbBaSaMo, 2024-04-25:
@shrayb once the outcome is determined, please make a comment in this issue and close it.

if there is still an issue and work to do, leave a comment but leave the issue open until it is resolved

AbBaSaMo, 2024-04-30:
new servo using ttl is on it's way back, closing issue now

closed 2024-04-30 by AbBaSaMo