Monash-Connected-Autonomous-Vehicle/ESDA #50: Install key dependencies on Hive5

AbBaSaMo, 2024-05-05:
- [ ] piksi deps
- [ ] ZCAM deps
- [ ] misc sudo packages
- [ ] install and build the current esda stack

AbBaSaMo, 2024-06-27:
VCS was not working. A file it tries to use requires root access. 

[image]

Used sudo with vcs then changed permissions of installed repos to user. @dylan-gonzalez @jaimasters what would be a better approach to solving this for future reference? Cause Ben was talking about the ills of slapping sudo on everything the other day. Idk if changing owner/group for /usr/local/lib/python3.10 onwards is appropriate so I avoided it.

dylan-gonzalez, 2024-06-27:
I didn't have that issue when I was installing ZED im pretty sure so idk

AbBaSaMo, 2024-07-01:
@dylan-gonzalez was speaking with Ben who was explaining that the issue was a result of running some installers with sudo and his suggested fix was chmodding necessary files to o=r-x

AbBaSaMo, 2024-07-01:
for personal ref: running `sudo chmod -R o=r-x /usr/local/lib/python3.10` has fixed it, it seems...

now onto the next issue
[image]

as per [this](https://community.stereolabs.com/t/build-failure-of-zed-ros2-wrapper-with-sdk-version-4-1/4997), the problem is that the latest version of zed sdk is not compatible with the ros2 wrapper. 

Going to try updating the fork so that new commits would hopefully fix it, otherwise, using a different version of the sdk that is compatible

AbBaSaMo, 2024-07-01:
Updating the fork seems to have fixed that. And now I have to deal with this
[image]

The issue is with the missing submodule. Deleted the empty directory on esda branch on our fork and now have no idea how to bring it back.

[image]
[image]

AbBaSaMo, 2024-07-01:
ok, done, the command was `git submodule add <zed_interface repo url> ./zed-ros2-interfaces`

AbBaSaMo, 2024-07-01:
yay it worked! closing issue for now

closed 2024-07-01 by AbBaSaMo