Monash-Connected-Autonomous-Vehicle/ESDA #56: Implement autonomous mode listenera/flags

AbBaSaMo, 2024-07-15:
### Overview
One key requirement for the competition is that the safety light must be on when the vehicle is on and blinking when the vehicle is in autonomous mode. Software defines autonomous mode as the duration of time where the vehicle is sending cmd_vel messages from Nav2 to the motors and the motors move in response to this. Thus so as long as cmd_vel is being sent by Nav2, this should somehow be communicated with the MCU chip via some boolean so that it can make the safety light blink.

### Acceptance criteria
1. While the vehicle is running and there are no cmd_vel messages being sent via ROS2, the safety light must be solid
2. While the vehicle is running and there are cmd_vel messages being sent, the safety light must be blinking

### Todo
- [ ] Communicate and determine design with electrical team
- [ ] Implement relevant code (checklist to be updated)

aallendarmawan, 2024-07-30:
This is handled by Electrical team so can this issue be closed?

AbBaSaMo, 2024-07-31:
I'll delete the issue instead.