Monash-Connected-Autonomous-Vehicle/ESDA #61: Nav2 configuration and research

AbBaSaMo, 2024-07-15:
### Overview
We will be using Nav2 to implement our path planning component. Nav2 is an open source stack that takes in a robots position via an `odometry` topic and can also take in a series of GPS waypoints and a map of occupied and unoccupied space before planning a path for the robot to take to reach the waypoints. This path is outputted as a series of `cmd_vel` messages that tell the robot what speed and direction to go to move through the waypoints.

Nav2 is a whole stack in and of itself and so requires various inputs and configurations beyond what was mentioned above. The gist of this task is to get it set up and working for our robot sow e can integrate it with the rest of our stack.

### Relevant resources
[Nav2 docs](https://docs.nav2.org/)

### Acceptance criteria
- [ ] Nav2 is set up and given a set of GPS waypoints, provides a series of `cmd_vel` messages for our robot to go through

### Todo
- [ ] Go through Nav2 docs and determine what is needed and what we currently have
- [ ] Look into URDF requirements
- [ ] Look into how to provide custom GPS waypoints
- [ ] Configure Nav2 for our robot and write a launch file for it
- [ ] Test it out while sensors are running but NOT control nodes

AbBaSaMo, 2024-07-20:
Note the following required transformshttps://docs.nav2.org/setup_guides/transformation/setup_transforms.html

aallendarmawan, 2024-07-30:
For GPS waypoints: https://docs.nav2.org/tutorials/docs/navigation2_with_gps.html?highlight=waypoint

aallendarmawan, 2024-07-30:
Our research: https://docs.google.com/document/d/1F9SW3ZAn4qdD49v2NXrQTKdig3iNO_7m6QgapqhZRzA/edit