Monash-Connected-Autonomous-Vehicle/ESDA #67: [DOCS] Complete ROSINFO.md

AbBaSaMo, 2024-08-11:
### Overview
ROSINFO documents all the topic names we use and the message types they output so that it is easy for members to understand what different components are doing. Thus, it is important to keep this up to date.

### Relevant resources
- [Our ROSINFO.md](https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/blob/main/ROSINFO.md)

Sample table detailing what is expected to be filled

[image]

### Acceptance criteria
- [ ] ROSINFO.md contains all relevant information for all node topics including source and objective

### Todo
- [ ] Either at regular intervals run integrated parts of the system and note rleevent info down into ROSINFO.md
- [ ] Or once all integration is completed do the above with the system running all at once.

nathan-sm, 2024-09-12:
@Jiawei-Liao I couldnt find the Alienware laptop so I had a crack at completing the task on the Beast. However after setting up ROS and searching for the topic list I could only find 'parameter_events' and 'rosout'. Even after trying to run individual packages/topics, terminal says the packages do not exist in the /opt/ros/humble tree. Note I am logged into mcav-fyp account as it was the only one available on the beast, not sure if there is another account im meant to be working out of.
Any help on how to properly set up ROS and the ros workspace so i can find relevant topics and packages would be great, and I can try get it done before the meeting today.

Jiawei-Liao, 2024-09-12:
The topics of 'parameter_events' and 'rosout' are just used for ROS itself to manage the systems. They appear even if you don't have any nodes running. 

Is the stack built on the beast? if so you might need to also source install/setup.bash. I'm not sure if the nodes in the stack work without the hardware plugged in.

If its not built, try following the steps in the readme.

nathan-sm, 2024-09-12:
Devices needed:
- Zed Camera
- LiDar 
- GPS

Jiawei-Liao, 2024-09-12:
I just realised that in the readme theres some instructions on how to setup the velodyne lidar:

Refer to the !(Velodyne LiDAR User Manual)[https://velodynelidar.com/wp-content/uploads/2019/12/63-9243-Rev-E-VLP-16-User-Manual.pdf] and/or !(YouTube setup guide)[https://www.youtube.com/watch?v=Pa-q5elS_nE] on further instructions on configuring the LiDAR

Hopefully these can help

nathan-sm, 2024-09-18:
Working session 18/09:
- Downloaded depencies for the zed driver, Swiftnav and Piksi Multi, aswell as VLP-16 LiDar
- Able to run the launch command for VLP-16 using command 'ros2 launch velodyne_driver velodyne_driver_node-VLP16-composed-launch.py' file, however launch stops executing midway.
- This launch file was found in the '/opt/ros/humble/share/velodyne_driver/launch' file path. Other launch files were in this dir.
- Colcon build ran successfully after all dependencies were installed
- When trying to launch esda_launch, FATAL error encountered '[rclcpp]: Called read in an unitialised SbpSerialDataSource'

Once we can get the packages to properly launch, it should be pretty straight froward to get the topic info including objctive and nodes interacting, however we cannot seem to set up the ROS environment correctly that allows for the packages to be launched. Would be good if we could work on this during Thursday meeting with someone who is more familiar with ROS and existing esda framework.

Yousouf-Palitanawala, 2024-09-18:
- Plugged in the Zed camera and the Velodyne V16 LIDAR, but could not access the Swiftnav Piksi multi

- Connected to the Zed camera, however with the LIDAR, were not able to access the local page with the data (tried 192.168.1.1 and didnt work) - set the IPv4 to manual and the address as 192.168.1.77 and netmask as 255.255.255.0 (based on Youtube video in README)

Jiawei-Liao, 2024-10-17:
ask UA @dtonda8 @ZileiChen on how to get the lidar to work

dtonda8, 2024-10-17:
Our setup for the twizy/van:
- Vehicle's computer & VLP 16 are connected to the same router via ethernet
- We've tweaked the VLP 16 settings and Wired settings as shown [here](https://github.com/dtonda8/twizy_sensor_kit_launch/blob/main/README.md)

To launch LiDar:
```bash
source /opt/ros/humble/setup.bash
ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py 
```

It is also possible to connect the VLP 16 directly to the computer. MCAV notion also has a [debug page](https://www.notion.so/monashcav/Debugging-Velodyne-VLP-16-LiDAR-ccac93d85e1748ebba3b9d2f12790870) for common errors (e.g. `Velodyne poll() timeout`)

I'll be on campus tomorrow evening and (maybe) next week Monday, dm me or Zilei if you need assistance and we can meet up.

Rikidink, 2024-10-18:
Got the web interface for the VLP 16 working. Was checking the data sent with wireshark and for some reason the route for the interface was 192.168.**10**.201 instead of the default 192.168.**1**.201. Solution is to just change the port address to match the subnet. So instead of setting the port to 192.168.**1**.77 (for example) you have to set it to 192.168.**10**.77:

```bash
# Assign IP to the ethernet port in 192.168.10.x range (assuming the ethernet interface is called eth0)
sudo ifconfig eth0 192.168.10.77

# Add static route to LIDAR's IP address:
sudo route add 192.168.10.201 dev eth0
```