Monash-Connected-Autonomous-Vehicle/ESDA #69: Find compute unit alternative

AbBaSaMo, 2024-08-11:
### Overview
The hive is good, but the hive is heavy.. and so elect and mechanical require that we find a more lightweight compute unit. A few alternatives exist, each with their own pros and cons and lead preferences so this task aims to get the highest preference working otherwise the next best and so on.

As for the alternatives, refer to the following table

| Pref | Alternative | Pros | Cons |
|------|-------------|------|-------|
| 2  | Buying a Jetson Orin Nano (not nano, ORIN nano) | | |
| 3  | Using our good old (old) Jetson TX2 |  | |

<table>
  <tbody>
    <tr>
      <th>Pref </th><th>Alternative </th><th>Pros</th><th>Cons</th>
    </tr>

    <tr>
      <td>1</td>
      <td>MCAV owned Alienware with ubuntu 22.04 install and ROS2 humble</td>
      <td>
        <ul>
        <li>item1</li>
        <li>item2</li>
        </ul>
      </td>

    </tr>

  </tbody>
</table>

### Relevant resources

### Acceptance criteria

### Todo

Rikidink, 2024-08-29:
Following this [forum post](https://askubuntu.com/questions/1454762/18-04-22-04-upgrade) to upgrade from Ubuntu 18.04 -> 22.04:
At the
sudo do-release-upgrade
Got "Could not calculate the upgrade" error seemingly because of incompatible/broken packages between 18.04 and 20.04.
Error in /var/log/dist-upgrade/main.log :
453 ERROR Dist-upgrade failed: 'E:Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.'
Resolved by [removing python-dev](https://askubuntu.com/questions/1279062/upgrade-from-18-04-to-20-04-prevented-by-eerror-pkgproblemresolverresolve-g):
sudo apt remove python-dev
sudo apt autoremove

Rikidink, 2024-09-02:
Finished update to 22.04.
Installing ESDA repo (and dependencies) but installing nvidia and cuda drivers as per ESDA repo readme breaks Ubuntu:
```
# install driver
sudo apt-get install -y nvidia-driver-550-open
sudo apt-get install -y cuda-drivers-550
```
Entered recovery mode and deleted drivers to get it in a bootable state again:
```
sudo apt-get purge nvidia*
```

AbBaSaMo, 2024-09-02:
@Rikidink this is the same issue we encountered with the other Alienware we have that is currently used for the Streetdrone project.

See if any of the suggested fixes below work:
- https://askubuntu.com/questions/955589/ubuntu-will-not-boot-normally-after-nvidia-driver-install
- https://askubuntu.com/questions/1129516/black-screen-at-boot-after-nvidia-driver-installation-on-ubuntu-18-04-2-lts
- https://forums.developer.nvidia.com/t/i-cannot-boot-ubuntu-22-04-after-installing-latest-nvidia-driver/251267/13

As a summary, some suggest using an older version of the driver, others suggest deleting certain files that somehow fix the issue. Once these and any other issues have been exhausted and we know it definitely won't work, let me know and we can close the issue until we get our hands on the orin nano.

Rikidink, 2024-09-07:
The Alienware can now run the stack (NOTE: used the sensor-integration branch of the ESDA repo). Closing issue temporarily until getting the Orin Nano to test on as well.

closed 2024-09-07 by Rikidink