Monash-Connected-Autonomous-Vehicle/ESDA #70: Demo night sim

AbBaSaMo, 2024-08-22:
### Overview
Demo night is in week 8. We want a hardware in the loop demo. This will involve:
- Running the simulation with Nav2 integrated
- Putting the ESDA on a stand
- Connecting it to the control syste,s
- Getting the drive wheels to mirror what the sim is doing

### Relevant resources
- Nav2 docs
- URDF docs

### Acceptance criteria
- [ ] URDF needs to have front steer wheels replaced with a frictionless caster
- [ ] Migrate to diff drive controller [can use gazebo plugin or ros2 control works case]
- [ ] Have a map of the IGVC course to spawn the robot inside
- [ ] Have Nav2 drive the robot to a user defined waypoint while basic lidar based obstacle detection and avoidance

### Todo
- [ ] Update URDF
- [ ] Include GNSS in URDF
- [x] Ask anthony where the map is
- [ ] Find the map in the part of the repo anthony pointed to: open each file in gazebo to see which on it is
- [ ] Integrate with Nav2 using appropriate dependency management and launching
- [ ] Test it and test it with Samuel Tri

AbBaSaMo, 2024-08-23:
I've updated the urdf to utilise casters as frictionless spheres. Included gazebo diff drive controller once again and added plugins to simulate IMU/GNSS in piksi.xacro obtained from https://medium.com/@bytesrobotics/a-review-of-the-ros2-urdf-gazebo-sensor-91e947c633d7.

Yet to test it but the only thing to really test is the IMU/GNSS component.

Jiawei-Liao, 2024-08-26:
I think the correct map/world is lino_track.yaml under /maps and world_only.model under /worlds. (I think world_only.model is the default world setting since it's used for tb3 sim)

[image]

The Gazebo doesn't show anything...

[image]

When I tried to change world_only.model to lino_map.world, it give me this repeatedly until it crashed.

[image]

Doesn't seem like an error? This is the final messages it printed before crashing/closing.

[image]

I think using world_only.model is fine? ChatGPT says it's for environment (lighting, physics). Gazebo didn't show anything so I can't verify this.

Not sure if it's my PC but I can't get the robot to spawn. I'm using default tb3 and I also tried with a fresh clone of the repo. All of them can load up the map but can't spawn the robot. Does it work for someone else?

AbBaSaMo, 2024-08-28:
@Jiawei-Liao when you do figure it out, rename the world to igvc and then whatever the extension is.

If we can't figure it out over time, we can always ask anthony

AbBaSaMo, 2024-08-28:
For my own sub-tasks, I have the following to do:

- [ ] Test diff drive with ''casters''
- [ ] Test piksi and imu pluging
- [ ] Integrate with slam toolbox and robot localisation

After this it can be integrated with nav2

AbBaSaMo, 2024-08-30:
idk if this fixed gazebo diff drive plugin or increasing the twist speed but it now works `sudo apt install ros-humble-gazebo-ros-pkgs`

AbBaSaMo, 2024-08-30:
@Monash-Connected-Autonomous-Vehicle/esda-team 

Quite janky when it moves. The "caster" wheels in the front have no friction so it shouuld be fine but it jumps around as shown in the video below.

Any ideas how to fix? I'll play around with inertia but all I can really do is update the mass until I find a sweet spot. Maybe position a single wheel at centre instead of 2.

https://github.com/user-attachments/assets/6e19b546-7d4b-4465-b5dd-2dc023d7f3a2

AbBaSaMo, 2024-08-30:
Moved casters forward a bit and it removed the lurching. Also added a rear caster, not sure if needed, can test without and the mass is not 0 but close to it

[image]

dylan-gonzalez, 2024-08-30:
The caster wheels aren't spinning, so changing the position of them won't fix anything.

Where have you set the friction to zero? I've done something like [this](https://github.com/Monash-Connected-Autonomous-Vehicle/adr/blob/d52f4efa8dcf971ba369b3bb36d7c0033ab95c13/adr_description/urdf/adr/adr.gazebo#L117-L135) for mine, I don't see anything like that in your urdf. (mine is for Gazebo Fortress so I'm not sure if the syntax is the same for classic but it probably is)

AbBaSaMo, 2024-08-30:
Oh, interesting. I used the same technique as josh newans in line 164 here where he basically set the mass to a very low number
https://github.com/joshnewans/articubot_one/blob/main/description/robot_core.xacro

The intertial_cylinder element in another file is where intertial physics is defined. In our repo it is under esda_sim/description/util

Problem was that without moving the casters forward, it jumps forward when the vehicle stops so I had to '''balance''' it if that makes sense. Will try out your solution when I'm next in.

Did you basically re-create the urdf in gazebo using sdf syntax??

dylan-gonzalez, 2024-08-30:
> Did you basically re-create the urdf in gazebo using sdf syntax??

The only SDF I have is the world file. Everything else is URDF - I use `<gazebo>` tags.

Jiawei-Liao, 2024-09-03:
I got the sim to run on the test world with tb3 bot. Small bot in such a big map. Movement works.

[image]
[image]

There is some output to /cmd_vel topic. Looks right I think? but not sure why only 1 value from linear/angular is being used at a time

[image]

Something that I noticed is that RVIZ laggs sometimes, resulting in the bot teleporting. But gazebo is smooth so I think it should be fine?

This is what I did to fix the issues I had before.
install:
`sudo apt install ros-humble-rmw-cyclonedds-cpp`

add to .bashrc:
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`

go to:
/opt/ros/humble/share/turtlebot3_navigation2/param

edit waffle.yaml (with sudo):
change: `robot_model_type: "differential"`
to: `robot_model_type: "nav2_amcl::DifferentialMotionModel`

then restart

This is the command used to run the sim:
`ros2 launch nav2_bringup esda_simulation_launch.py`

To change the map, change the maps and worlds variables
It is currently using nav2_params instead of esda_params. I don't think we need to change this for the demo since I don't think we need anything else (custom layers/plugin)
The model is waffle but it can be changed with robot_name and robot_sdf variables.

dylan-gonzalez, 2024-09-03:
> but not sure why only 1 value from linear/angular is being used at a time

what do you mean? linear x is forward, angular z is yaw. Can your bot move sideways, or fly in the air? Otherwise these values make sense

> Something that I noticed is that RVIZ laggs sometimes

Can you be specific? Do you mean one of the transforms is lagging/glitching?

Jiawei-Liao, 2024-09-03:
> what do you mean? linear x is forward, angular z is yaw. Can your bot move sideways, or fly in the air? Otherwise these values make sense
Ah I see, that makes sense. I thought it would use a combination of linear x, y, z. But I guess the bot is just moving in it's x direction, not the RVIZ axis.

> Can you be specific? Do you mean one of the transforms is lagging/glitching?

https://github.com/user-attachments/assets/2c000e79-6d0b-49fa-bfe2-db73ecbde900

dylan-gonzalez, 2024-09-03:
Yeah so that's something to do with your localisation. You're running AMCL right? So maybe look at some forums related to that, probs have to tune parameters or something.

But everything is working fine it seems in terms of gazebo and rviz

AbBaSaMo, 2024-09-03:
@Jiawei-Liao did i miss something or is there a reason the test world does not have the ground with the grass and lanes?

Jiawei-Liao, 2024-09-04:
> [@Jiawei-Liao](https://github.com/Jiawei-Liao) did i miss something or is there a reason the test world does not have the ground with the grass and lanes?

oh is that not the map? The other worlds/maps are all the same as this. I'll ask anthony on this. Is he still on slack?

AbBaSaMo, 2024-09-04:
It's the right map but it doesn't seem to have loaded properly. Idk if u remember but when Anthony was showcasing it, it had a grass plane with a track.

He's on slack but idk if he's in the team so might be best to shoot him an email.

AbBaSaMo, 2024-09-05:
- [ ] add nav2 to package.xml in launch pkg
- [ ] copy over the 2 map files and any configs to esda_sim
- [ ] create the launch file while pointing to relevant urdfs/,ap files and configs

Jokua, 2024-09-09:
Just writing a guide on how to get the simulator to run with the basic stuff from the Nav2 Tutorial after Jiawei's fixes:

`source /opt/ros/humble/setup.bash`
`export TURTLEBOT3_MODEL=waffle`
`export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models`
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
`ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`

After it launches gazebo and rviz2, you have to set the 2D pose estimate of the robot. You can do that by guess-timating the spot by looking at the gazebo sim.

In order to set the goal pose, don't forget to click, hold and aim it in the direction you want it to face at the goal. You can click the goal but it's better if you set the goal pose facing where you want it to to show that it can compensate for that or something

AbBaSaMo, 2024-09-10:
Need the following links in the URDF
- [x] base_link
- [x] base_footprint

Regarding the odom -> base_link tf, using robot_localisation
- [x] output GPS from gps plugin as /navsatfix
- [x] output imu as /imu
- [x] output wheel odom from gazebo diff drive plugin as /wheel_odom
- [x] configure robot_localization and verify the transforms

AbBaSaMo, 2024-09-11:
Now slam tool box is working witht he URDF. I changed the ROS DOMAIN ID to 10 as i was getting transforms not defined in my urdf, and it seemed after some time the map was empty but then began publishing fine

[image]

Now on to the last step: Nav2 integration :D