Monash-Connected-Autonomous-Vehicle/ESDA #72: URDF for sim set up

AbBaSaMo, 2024-09-02:
For my own sub-tasks, I have the following to do:

- [x] Test diff drive with ''casters''
- [x] Test piksi and imu pluging
- [ ] Integrate with slam toolbox and robot localisation

After this it can be integrated with nav2

_Originally posted by @AbBaSaMo in https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/issues/70#issuecomment-2314386359_

Jiawei-Liao, 2024-09-03:
@dylan-gonzalez do you know what is going on here?

https://drive.google.com/file/d/12ioet2Uh13NLNTiPKz3FtzoDEJdI4z3r/view?usp=sharing

I was following the get started guide (https://docs.nav2.org/getting_started/index.html) just to get something working. At least it starts up and spawns the turtle bot. I did notice that in RVIZ, the bot spawns in the middle but in gazebo it spawns in the normal corner. I can give it the pose estimate and goal but then it goes crazy. In gazebo, the bot can move a little bit but is basically stuck. 

The only part I didn't follow the guide is creating transforms from baselink, odom and map. But it seems like it needs it, otherwise it can't run.

I have no idea whats going on or how to fix it...

dylan-gonzalez, 2024-09-03:
You need to set up the transforms otherwise the bot won't know where it is

Jiawei-Liao, 2024-09-03:
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link odom
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link map

I just ran these 2 commands in another window. Or is there something else that I'm missing?

dylan-gonzalez, 2024-09-03:
yeah but that's just a static transform base_link -> odom, so when you set a goal pose the robot won't be able to update it's position as it's trying to move toward the goal.

Isn't there a diff drive odometry you can use to give you base_link -> odom? and then you can set a static transform odom -> map

_also probably the issue that the robot is glitching about is because you have two parents for base_link... typically your tf tree should look like base_link -> odom -> map_

Jiawei-Liao, 2024-09-03:
Ohh that makes sense. Thanks!