Monash-Connected-Autonomous-Vehicle/ESDA #84: Integration of LiDAR and Piksi GPS

Rikidink, 2025-04-07:
Try and integrate the LiDAR and the Piksi GPS into our current stack.
Specifically, we need integration with ROS2 and then down the track, into NAV2.

Refer to this previously closed issue for more information on the VLP16 LiDAR #21 

### Task:
- [ ] Get the actual LiDAR connected via ethernet, you can try following the YouTube [video](https://www.youtube.com/watch?v=Pa-q5elS_nE) although I believe the IP address of the LiDAR was modified at some point so refer to Dylan's [comment](https://github.com/Monash-Connected-Autonomous-Vehicle/ESDA/issues/21#issuecomment-1905492567) in #21
  - Optionally, you can download [VeloView](https://www.paraview.org/veloview/) to see the data (but not really important)
- [ ] Integrate the LiDAR with ROS2: again, try to refer to #21 and/or online resources
- [ ] Research integration with NAV2
---
Similar situation with the Piksi GPS:
- [ ] Get the Piksi configured and setup according to the README
- [ ] Research ROS2 integration and then with NAV2

yakuraku, 2025-04-21:
## UPDATE: SwiftNav ROS2 Driver Build Issue

After 6 hours of troubleshooting the SwiftNav ROS2 driver build issues in my Humble environment, I've identified several problems:

1. The primary error occurs in the `swiftnav_ros2_driver` package which fails with:
   ```
   CMake Error at CMakeLists.txt:41 (find_package):
     By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
     has asked CMake to find a package configuration file provided by
     "ament_cmake", but CMake did not find one.
   ```

2. There's also an upstream issue with the `libsbp` Python package:
  ```
   RuntimeError: 'distutils.core.setup()' was never called -- perhaps 'setup.py' is not a Distutils setup script?
  ```

I've confirmed this is not a general ROS2 Humble build environment issue as other ROS2 packages build successfully in the same workspace. The problem appears to be specific to the SwiftNav driver and its dependencies.

My next steps are:
- Bypass the problematic libsbp package using COLCON_IGNORE
- Focus on building just the swiftnav_ros2_driver with proper ROS2 Humble dependencies
- Check if there are compatibility issues requiring patches for Humble

Will update as I make progress.

Rikidink, 2025-04-24:
# Lidar working with ROS2 with data being shown in RVIZ

Commands are derived from this tutorial:
https://yavuzertugrul.com/Lidar_part_1

### Routing and web interface:
To get access to the web interface:
(assuming network interface called `enP8p1s0`, if not, check ethernet interface using `ifconfig`)
NOTE: in online resources/tutorials `192.168.1.201` is the default network/sensor IP but we're actually using `192.168.10.201`
`sudo ifconfig enP8p1s0 192.168.10.201`
`sudo route add 192.168.10.201 enP8p1s0`
Set netmask to 255.255.255.0 & Gateway to 0.0.0.0 in the wired network options in Ubuntu/Operating system settings
Should now be able to see the web interface by opening a web browser and going to `192.168.10.201`

### ROS2 and RVIZ
(assuming ROS2 Humble and RVIZ is installed)
```
  mkdir -p ~/ros2-ws/src
  cd ~/ros2-ws/src
  git clone -b humble-devel https://github.com/ros-drivers/velodyne.git
  cd ~/ros2-ws
  rosdep install --from-paths src --ignore-src -y
  colcon build
  source install/setup.bash #also if you don't want to do this everytime source it in bash
```
Since the network/sensor IP is at `192.168.10.201`  we need to change the velodyne_driver config to recognise this:
```
# in ros2_ws directory
cd src/velodyne/velodyne_driver/config/
nano VLP16-velodyne_driver_node-params.yaml
```
Change the `device_ip` field to `192.168.10.201` and save the file
```
cd ../../../..
source install/setup.bash
colcon build
```
#### Launch:
```
ros2 launch velodyne_driver velodyne_driver_node-VLP16-launch.py

# In a separate terminal:
ros2 launch velodyne_pointcloud velodyne_transform_node-VLP16-launch.py

# In a separate terminal:
ros2 topic list
# You should be able to see the velodyne topics after listing
ros2 topic echo velodyne_points
# After running above command you should be able to see the data 
# CTRL+C and then run:
ros2 run rviz2 rviz2 -f velodyne
```

The RVIZ window should now be open. To see the pointcloud data:
1. Press `Add` in the bottom left corner
2. Press the `By topic` tab
3. Select `PointCloud2` under /velodyne_points and press `Ok`
4. You should now be able to see visual data from the VLP16

Rikidink, 2025-05-18:
Voxel filter

```cpp
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/point_cloud2.hpp>
#include <pcl_conversions/pcl_conversions.h>
#include <pcl/filters/voxel_grid.h>
#include <pcl/point_cloud.h>
#include <pcl/PCLPointCloud2.h>
#include <pcl/io/pcd_io.h>

using std::placeholders::_1;

class VoxelFilterNode: public rclcpp::Node
{
    public:
        VoxelFilterNode() : Node('voxel_filter_node')
        {
            subscribe_ = this->create_subscription<sensor_msgs::msg::PointCloud2>(
                // this might be different when using the VLP32
                "/velodyne_points", rclcpp::SensorDataQoS(),
                std::bind(&VoxelFilterNode::pointCloudCallback, this, _1));

            // published to /velodyne_voxel_filtered ROS2 topic
            publish_ = this->create_publisher<sensor_msgs::msg::PointCloud2>('/velodyne_voxel_filtered', 10);
            
            RCLCPP_INFO(this->get_logger(), "Voxel filter node started.");
        }
    
    private:
        void pointCloudCallback(const sensor_msgs::msg::PointCloud2::SharedPtr msg)
        {
            // Create new PCL PC2 and convert Velodyne data into PCL format
            pcl::PCLPointCloud2::Ptr pcl_pc2 (new pcl::PCLPointCloud2());
            pcl_conversions::toPCL(*msg, *pcl_pc2);

            // Filtered PC2 
            pcl::PCLPointCloud2::Ptr voxel_filtered (new pcl::PCLPointCloud2());

            // (Debug) PC points before filter
            std::cout<<"Source Cloud Points "<< pcl_pc2->width * pcl_pc2->height<< std::endl;

            // Create filtering object
            pcl::VoxelGrid<pcl::PCLPointCloud2> voxel_filter;
            voxel_filter.setInputCloud(pcl_pc2);
            voxel_filter.setLeafSize(0.01f, 0.01f, 0.01f); // leaf size (adjust as needed for desired downsampling)
            voxel_filter.filter(*voxel_filtered);

            // (Debug) PC points after filter
            std::cout<<"Voxel Cloud Points "<< voxel_cloud->width * voxel_cloud->height<< std::endl;

            // Convert back to ROS2/PointCloud2 sensor msg
            sensor_msgs::msg::PointCloud2 output;
            pcl_conversions::fromPCL(*voxel_filtered, output);
            output.header = msg->header; // idk what this does

            publish_->publish(output);
        }
        
        
        rclcpp::Subscription<sensor_msgs::msg::PointCloud2>::SharedPtr sub_;
        rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr pub_;
}
```

Rikidink, 2025-05-18:
## voxel_filter.cpp

```cpp
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/point_cloud2.hpp>
#include <pcl_conversions/pcl_conversions.h>
#include <pcl/filters/voxel_grid.h>
#include <pcl/point_cloud.h>
#include <pcl/PCLPointCloud2.h>
#include <pcl/io/pcd_io.h>

using std::placeholders::_1;

class VoxelFilterNode: public rclcpp::Node
{
    public:
        VoxelFilterNode() : Node("voxel_filter_node")
        {
            subscribe_ = this->create_subscription<sensor_msgs::msg::PointCloud2>(
                // this might be different when using the VLP32
                "/velodyne_points", rclcpp::SensorDataQoS(),
                std::bind(&VoxelFilterNode::pointCloudCallback, this, _1));

            // published to /velodyne_voxel_filtered ROS2 topic
            publish_ = this->create_publisher<sensor_msgs::msg::PointCloud2>("/velodyne_voxel_filtered", 10);
            
            RCLCPP_INFO(this->get_logger(), "Voxel filter node started.");
        }
    
    private:
        void pointCloudCallback(const sensor_msgs::msg::PointCloud2::SharedPtr msg)
        {
            // Create new PCL PC2 and convert Velodyne data into PCL format
            pcl::PCLPointCloud2::Ptr pcl_pc2 (new pcl::PCLPointCloud2());
            pcl_conversions::toPCL(*msg, *pcl_pc2);

            // Filtered PC2 
            pcl::PCLPointCloud2::Ptr voxel_filtered (new pcl::PCLPointCloud2());

            // (Debug) PC points before filter
            std::cout<<"Source Cloud Points "<< pcl_pc2->width * pcl_pc2->height<< std::endl;

            // Create filtering object
            pcl::VoxelGrid<pcl::PCLPointCloud2> voxel_filter;
            voxel_filter.setInputCloud(pcl_pc2);
            voxel_filter.setLeafSize(0.05f, 0.05f, 0.05f); // leaf size (adjust as needed for desired downsampling, )
            voxel_filter.filter(*voxel_filtered);

            // (Debug) PC points after filter
            std::cout<<"Voxel Cloud Points "<< voxel_filtered->width * voxel_filtered->height<< std::endl;

            // Convert back to ROS2/PointCloud2 sensor msg
            sensor_msgs::msg::PointCloud2 output;
            pcl_conversions::fromPCL(*voxel_filtered, output);
            output.header = msg->header; // idk what this does

            publish_->publish(output);
        }
        
        
        rclcpp::Subscription<sensor_msgs::msg::PointCloud2>::SharedPtr subscribe_;
        rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr publish_;
};

// main function
int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<VoxelFilterNode>());
    rclcpp::shutdown();
    return 0;
}
```

## CMakeLists.txt

```c
cmake_minimum_required(VERSION 3.8)
project(voxel_filter)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# Find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(PCL REQUIRED)
find_package(pcl_conversions REQUIRED)

include_directories(
  include
  ${PCL_INCLUDE_DIRS}
)

add_executable(voxel_filter_node src/voxel_filter.cpp)

ament_target_dependencies(voxel_filter_node
  rclcpp
  sensor_msgs
  pcl_conversions
)

target_link_libraries(voxel_filter_node
  ${PCL_LIBRARIES}
)

install(TARGETS
  voxel_filter_node
  DESTINATION lib/${PROJECT_NAME}
)

ament_package()

```